equinor/flotilla #798: Implement caching in frontend

GodVenn, 2023-05-11:
**Describe the new feature you would like to see**
We should cache the responses from the backend to not "lose" the information each time we switch pages.
Also some things, like the echo missions maybe, could be cached until page refresh (session cache)

**Describe the solution you'd like**
There are several ways to do this, this article describes some 
https://betterprogramming.pub/exploring-caching-techniques-in-react-d30bbb78d54d

**How will this feature affect the current Threat Model?**
N/A

pooranjoyb, 2023-05-12:
Would be glad if you'd assign me this issue. @GodVenn @Eddasol

GodVenn, 2023-05-12:
Sure, we are happy yo take contributions!

closed 2023-09-20 by andchiind