equinor/flotilla #808: Frontend components does not adjust to dynamic window size

anetteu, 2023-05-21:
**Describe the bug**
Frontend components does not adjust to dynamic window size. 

**To Reproduce**
Open flotilla in a browser and change the window size. 

**Expected behavior**
Frontend components should adjust to window size.

**Screenshots**
Screenshots from iPad, zoomed out on the second screen shot. Several frontend components does not adjust to window size. 
[image]
[image]

mrica-equinor, 2023-06-02:
Had a look at this with Edda and there is no good solution Since each card is generated individually and there are different missions and robots, to have flexible adjustment to the window size could present the information in a confusing way and not aligned. For now we believe it is better to leave it as it is and people can scroll to the right if they need to see more information.

aestene, 2023-08-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-24:
Why is this issue archived from board but not closed? Should either be closed or open and on the board. Restoring issue and moving to new to force an evaluation during planning.

oysand, 2023-09-25:
Should be closed