equinor/flotilla #811: Modify assetDecks to contain a list of possible localization points

Eddasol, 2023-05-25:
**Describe the improvement you would like to see**
Change DefaultLocalizationPose in AssetDeck to be a list

**How will this change existing functionality?**
It will be possible to have larger decks, meaning less need to update currentDeck and possibility to choose between more safesones.

**How will this improvement affect the current Threat Model?**
N/A

Eddasol, 2023-05-25:
Should probably not be done before portals are functioning (to make sure that the robot chooses a safe zone that's possible to reach)

Also note that localization part uses this value and needs to be adapted to handle more points

aestene, 2023-06-19:
Not required if we are going for #852 

Until we do so we should be good with one localization point per deck. If required we can re-open this issue.

closed 2023-06-19 by aestene