equinor/flotilla #833: WIP: Add feature for supporting inspection plans

aestene, 2023-06-05:
**Describe the new feature you would like to see**
Flotilla will support regular inspections of different types of equipment. Most equipment have an inspection interval which indicates how often they are required to be inspected. Using robotics this interval may be shortened, but Flotilla needs to maintain an overview of how long it has been since a mission was executed, which interval it should be executed at and when is the next time it should be executed.

With this is the requirement for a visualization which will give operators an overview of which areas have missions that should be scheduled soon. 

Requirements:
- Missions should have an optional interval which may be specified
- Missions should be associated with asset decks
- There should be an indication for an asset deck on how long it has been since missions has been run on this deck and which missions are required to be run (note that different missions may have different intervals)

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2023-07-13:
The first two requirements will be covered in https://github.com/equinor/flotilla/pull/864. The last point can be done by filtering mission definitions by area and then comparing the inspection frequency with the date of the last run.

oysand, 2023-09-25:
Final step is to verify and promote to production

aestene, 2023-11-09:
Implemented

closed 2023-11-09 by aestene