equinor/flotilla #850: Localization map needs a heading

Christdej, 2023-06-14:
When localizing the robot, there is no indication which way the robot should be facing. We should change the dot in the figure from a circle with the number 1 in it, to the same we use for the robot in the normal map, but with the heading the localization is expecting.

[image]
[image]

Aswin-Inigo, 2023-06-15:
asdf

Eddasol, 2023-06-15:
[image]

Would this be preferred? @Christdej

Christdej, 2023-06-15:
That would be perfect yes! @Eddasol

closed 2023-06-16 by Eddasol