equinor/flotilla #854: Add "source" to "missionDefinition"

Eddasol, 2023-06-14:
**Describe the improvement you would like to see**
Switch out echoMissionId in missionDefinition with an object called source. 
[image]

**How will this change existing functionality?**
It be used to make it possible to choose between missions from different sources

**How will this improvement affect the current Threat Model?**

aestene, 2023-09-12:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-24:
Why is this issue archived from the board but not closed? Should either be closed or open and on the board. Restoring issue and moving to new to force an evalution during planning.

oysand, 2023-09-25:
Should be closed

closed 2023-09-25 by oysand