equinor/flotilla #875: Add Filter and better navigation to Mission History

OleDrange, 2023-07-03:
**Describe the improvement you would like to see**
Add Filter and better navigation to Mission History. Now it looks like the image below. Should be able to navigate quickly to page 30, or any other page. Now this takes 20 clicks at least. 
Also, it would be good to go back to the same place after pressing back in a mission. Now this will return you to page 1 even if you clicked the mission on page 30. 
[image]

**How will this change existing functionality?**
Possible to find missions easier in the mission history. 

**How will this improvement affect the current Threat Model?**

aestene, 2023-09-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-24:
Why is this issue archived but not closed? I think this is implemented? Restoring issue and moving to new to force an evalution during planning.

oysand, 2023-09-25:
This should be closed

closed 2023-09-25 by oysand