equinor/flotilla #876: Should use plantCode when fetching echo missions

Eddasol, 2023-07-03:
Using installation code when fetching missions from echo gives a lot more missions than the once that are relevant. Will e.g. get missons from both Haugsneset and Johan Sverdrup when choosing either one of them. Same with Kårstø and Kalstø. See /api/plantInfo in https://echohubapi.equinor.com/.

Suggest getting echo to change parameter in endpoint "/api/robots/robot-plan" from installation code to plant code and implementing the same change on our side

andchiind, 2023-08-01:
The necessary changes on our end have been implemented in #926 but the changes are not in Echo at the moment

aestene, 2023-10-01:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2024-04-29:
It is possible to amend the UI to make it display Installations instead of all the Plants

closed 2025-10-27 by aeshub