equinor/flotilla #880: Localization of Robot doesn't work on Chrome

Afonso-2403, 2023-07-05:
**Describe the bug**
When trying to localize the robot on chrome on an mobile device the screen goes blank

**To Reproduce**
In flotilla on an Android tablet:
Select a robot. 
Press Localize robot. 
Select a deck. 
The screens goes blank. 

**Expected behavior**
Same as in edge or on a PC, after selecting a deck a map of the deck is showed with the location where the robot should be placed for localization.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Eddasol, 2023-07-12:
None of the autocomplete components work on chrome which probably is the reason for this issue.

The same issue occur when viewing as pad through inspect

oysand, 2023-08-08:
Might not be relevant after #950

closed 2023-08-08 by oysand

Eddasol, 2023-08-08:
Autocomplete components are used for other thing as well, like selecting asset and scheduling missions. This is still relevant for those, but the issue could be rewritten