equinor/flotilla #897: Consider adding option of add mission from history on front page

anetteu, 2023-07-19:
**Describe the new feature you would like to see**
Consider adding option of adding/re-run missions from history together with the other add option button just below the mission queue. With reference to the screenshot, a suggestion could be to replace the ` + Add mission` button with the two buttons  `+ Add mission from echo` and `+ Add mission from history`. Another solution could be to keep the `+ Add mission` button as is, and letting that button open a menu where the user chooses the source to add mission from. 

The functionality of adding/re-running mission from history is already available on the history page, and I suggestion keeping that functionality as is, in adding the suggested option.

[image]

**Describe the solution you'd like**
If you have an idea of how to implement the feature please describe it here.

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-09-23:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-10-02:
Awaiting to see if this comes as a request through user involvement. We'll try and have Inspection page as the primary way to schedule the missions

closed 2023-10-02 by oysand