equinor/flotilla #899: Improve error message for arm movements

anetteu, 2023-07-19:
**Describe the improvement you would like to see**
I suggest making to changes to the current display of the error message that appears when an arm movement is scheduled while the robot is already busy with an ongoing arm movement:

1. Display the error message on a red pop up window, as is done for other error messages.
2. Rewrite the error message from the error code to something more understandable for the user, along the lines "Could not start the requested arm movement as the robot is busy with an ongoing arm movement"

An even better solution is to make the buttons not clickable when the arm is busy. 

[image]

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

anetteu, 2023-07-23:
Could also consider having a toggle button for batterychange/transport. Based on observations from users using the interface.

aestene, 2023-09-21:
This issue has automatically been marked as stale as there has been no activity for 60 days.

anetteu, 2023-09-23:
The buttons are now unavailable during arm operation, hence this issue on error messages can be closed. A smiliar issue for status messages are opened as #1024

closed 2023-09-23 by anetteu