equinor/flotilla #900: Re-arrange/cleanup the status fields on the ongoing mission page

anetteu, 2023-07-19:
**Describe the improvement you would like to see**
At the moment, there is a lot of text and different fields next to each other on the ongoing mission page which makes it a bit messy and not very readable. Consider re-designing or cleanup this section. Please see the attached screenshot.

[image]

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2023-07-24:
Need to consult UX design

aestene, 2023-09-23:
This issue has automatically been marked as stale as there has been no activity for 60 days.

UsamaEquinorAFK, 2024-01-09:
Link to Figma: https://www.figma.com/file/YoZo53WJmjlrjdQmJSdoHX/Flotilla---Sandbox?type=design&node-id=1%3A57745&mode=dev&t=CqblRNgINyWjnz8I-1

PR #1326 (2024-01-12): Refactor front page for iphone use
Closes #900
PR #1328 (2024-01-12): Refactor ongoing mission page for iphone use
Closes #900
closed 2024-01-12 by mrica-equinor