equinor/flotilla #902: Go to home mission always last in mission queue

anetteu, 2023-07-20:
**Describe the new feature you would like to see**
A go to home/safe position mission that is always scheduled as the last mission of the mission queue if the mission queue is not empty, or there is an ongoing mission. The mission should always be the last one in the queue, even if new missions are scheduled. The desired behavior of the robot when an ongoing mission is completed will then be: if mission queue not empty -> run next mission. else -> go to home/safe position.

**Describe the solution you'd like**
If you have an idea of how to implement the feature please describe it here.

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2023-07-24:
It could be nice to make it easier to schedule a go to safe position mission, but I am not sure if we always want this to be queued up by default. For instance if a mission fails and we quickly want to rerun it.

oysand, 2023-07-31:
Should have a larger discussion about return home functionality before jumping in to it

anetteu, 2023-08-02:
Related to #851 #921. If a discussion is needed here, it should start with a definition of the intended end user. Are we devolving a product to be used by dev team members, or by field operators?

oysand, 2023-08-08:
To be handled in #950

closed 2023-08-08 by oysand