equinor/flotilla #908: Consider improving the apperance on small touch screen/iPhone

anetteu, 2023-07-23:
**Describe the improvement you would like to see**
The frontend design and components are not all optimized from small touch screens/iPhones. Not all components are easy to adjust with the current deisgn as they have a lot of content. It should, however, be considered to make the needed changes in design and implementation to improve the user experience on small touch screens/iPhones. An example is adjustment of the header menu. This new issue is created to represent the need of a good user experience on iPhones. Some related issues are: #808, #906.

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

andchiind, 2023-07-24:
Here we need to involve UX design

anetteu, 2023-09-18:
#1009

closed 2024-01-22 by tsundvoll