equinor/flotilla #909: Current task number on ongoing mission card shows number of completed tasks, not ongoing task number

anetteu, 2023-07-23:
**Describe the bug**
On the ongoing mission card on the front page, there is a task counter `Task 1/3` that shows what I understand as the ongoing task number. It does, however, shows the number of completed tasks. This means that it shows `Task 0/3` during execution of task 1. Either it shows the wrong number, or it is confusing what the number should be. 

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

andchiind, 2023-07-24:
The main issue may be the ordering of the mission tasks themselves. If we fix this then this is no longer an issue.

anetteu, 2023-07-27:
> The main issue may be the ordering of the mission tasks themselves. If we fix this then this is no longer an issue.

I am not sure if there is an issue here, I am just questioning what this counter should show. At the moment it is implemented to show number of completed tasks. I suggest to change the implementation to show the task number of the ongoing task, as that is how it is understand by several of the users.

aestene, 2023-09-26:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-10-02:
Consider just calling it completed tasks to avoid ambiguity? (Given that there is enough space

closed 2023-11-09 by anetteu