equinor/flotilla #912: Make battery and pressure status status grey when robot is offline

anetteu, 2023-07-25:
**Describe the improvement you would like to see**
A suggestion to make it visible that the values are not live.

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

UsamaEquinorAFK, 2023-07-31:
Is it compliant with the design we have in Miro ?

UsamaEquinorAFK, 2023-07-31:
@mariannewiggen

mariannewiggen, 2023-07-31:
I think it makes sense to do this if the status of the robot is offline. If the status of the robot is not offline, then I don't think it is clear enough that battery and pressure change color due to values not being live.

Link to suggestion for grey battery and pressure status in Figma: https://www.figma.com/file/q6vsNRogzV6r9vTCCDsocd/Robot-Missions?type=design&node-id=4611%3A594648&mode=design&t=CSQq7Htipx2CzkLf-1

[image]

mariannewiggen, 2023-07-31:
Another suggestion could be to not show the value of battery and pressure if there is a high probability that the values will be wrong

[image]

UsamaEquinorAFK, 2023-08-01:
I prefer the second option. Thanks :)

UsamaEquinorAFK, 2023-08-02:
#930

closed 2023-08-03 by UsamaEquinorAFK