equinor/flotilla #918: Improve the re-run symbol on a historic mission page

anetteu, 2023-07-27:
**Describe the improvement you would like to see**
Based on observation of several demos, it is not obvious that the arrow on the historic mission page will re-run a mission. It should be considered to make a button with text or in other way make it more intuitive. 

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-12-18:
Did you look into this @Eddasol or @mrica-equinor ?

Eddasol, 2024-05-02:
[image]
Design from figma

Eddasol, 2024-07-01:
Need to update button design to match figma sketch

PR #1647 (2024-07-01): Change rerun mission button
Closes #918
PR #1663 (2024-07-02): Change rerun buttons
Closes #918
closed 2024-07-02 by haakonsf