equinor/flotilla #928: Fix dropdown to select asset on frontend

mrica-equinor, 2023-08-02:
**Describe the bug**
Sometimes when the flotilla page is refreshed the dropdown to select the asset appears to the left of the screen and it happens quite a lot that the selected asset changes and the robot disappears. 

We have talked briefly about creating a page before the frontpage where the asset is selected and therefore can't be changed by miss-clicking somewhere on the frontpage. Something similar to the echo page where you select the asset would be my suggestion for this. I believe this is a good solution since most of the operators will be using flotilla only for the asset they are operating on.

Another option is to just fix what causes the dropdown to re-render when the page is refreshed, and only when the down arrow by the right of the dropdown is pressed.

**To Reproduce**
Refresh the flotilla front page, especially with the mouse in the top of the page, for example by the flotilla logo.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

Christdej, 2023-08-08:
I think also, like in Echo, that we should remeber the last facility that user used. Gives a smoother user experience, as 99% of user will only be on one facility.

aestene, 2023-08-15:
@mrica-equinor Can you confirm whether this issue still occurs after #971 ?

If the issue should remain to implement a new asset selection page please rewrite it to focus on that.

andchiind, 2023-10-12:
The problem no longer seems to be relevant with the new starting page

closed 2023-10-12 by andchiind