equinor/flotilla #931: The mission is not correctly "In Progress" during the first task

Christdej, 2023-08-02:
**Describe the bug**
When starting a mission the first task is never in the state "In Progress" and hence the start / stop / pause buttons are disabled. See picture one. Then when the robot has completed the first task, the first task is put to "Successful" without entering "In Progress", the second task is correctly going to "In Progress" and the "Stop / Pause" buttons are back.

**To Reproduce**
Steps to reproduce the behavior:
Go to historic missions and re-run a previous mission

**Screenshots**
[image]
[image]

anetteu, 2023-08-02:
Covered by #896 #887

anetteu, 2023-08-15:
According to statement in #941 and #896, the stop and pause button is now available through the entire mission.

andchiind, 2023-09-20:
@Christdej can we close this? Since the PRs linked here are closed?

anetteu, 2023-09-23:
This issue is a duplicate of #887, which is closed since it is covered by this issue. The issue is still present in prod, so unless it is fixed since the last release we should not close this issue.

aestene, 2023-12-18:
Solved

closed 2023-12-18 by aestene