equinor/flotilla #932: Fix/redesign robot page for when its offline

UsamaEquinorAFK, 2023-08-03:
Right now, if the Robot is offline, the robot card is still "clickable" and the user can access the robot page. 

1- Should it be accessible ? 
2- If yes, Should it have the localize button visible on an offline robot ? 
3- If yes to both above, what should the robot page for an offline robot look like in general - do we have a design on figma or is it something we shouild discuss over ?

UsamaEquinorAFK, 2023-08-03:
<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAdEAAAJpCAIAAAB90J9JAAAgAElEQVR4AeydCVxU1RfHE7fMtM0FBVyzxfZ9+YdapEVZ2qKVWqa5YaaVWaZWLqmJiXtuuKCACLKpIKvszD7zZt/3mTf7sC+DKP/PfQ8QEBHQaBjPfN4H3sy89+6933PnN2fOPfe+u1zwAAJAAAgAga4icFcdPIAAEAACQKCrCIDmdhVpKAcIAAEgUFcHmgu9AAgAASDQdQRAc7uONZQEBIAAEADNhT4ABIAAEOg6AqC5XccaSgICQAAIgOZCHwACQAAIdB0B0NyuYw0lAQEgAARAc6EPAAEgAAS6jgBobtexhpKAABAAAqC50AeAABAAAl1HADS361hDSUAACAAB0FzoA0AACACBriMAmtt1rKEkIAAEgABoLvQBIAAEgEDXEQDN7TrWUBIQAAJAADQX+gAQAAJAoOsIgOZ2HWsoCQgAASAAmgt9AAgAASDQdQRAc7uONZQEBIAAEADNhT4ABIAAEOg6AqC5XccaSgICQAAIgOZCHwACQAAIdB0B0NyuYw0lAQEgAARAc6EPAAEgAAS6jgBobtexhpKAABAAAqC50AeAABAAAl1HADS361hDSUAACAAB0FzoA0AACACBriMAmtt1rKEkIAAEgABoLvQBIAAEgEDXEQDN7TrWUBIQAAJAADQX+gAQAAJAoOsIgOZ2HWsoCQgAASAAmgt9AAgAASDQdQRAc7uONZQEBIAAEADNhT4ABIAAEOg6AqC5XccaSgICQAAIgOZCHwACQAAIdB0B0NyuYw0lAQEgAARAc6EPAAEgAAS6jgBobtexhpKAABAAAqC50AeAABAAAl1HoLtqrtrh7BikGu6sn37ts2J1nxWrB65c/+Tus0e1lR27AnF0YfK+txPVnTgRTgECQAAI1NXVdUvNfWvv4bEbtnVMdqu5H/+0fk6OulCjLRRhfx7YMnBd5KWqqx3sBJdjT/z+1FnQ3A5ig8OBABBoIND9NPetvYf7rFh9gsZsaEL7/iPN3bxKdqX+aEPG8yv37DCC5raPHhwFBIDAbSLQzTS3k4JbV1dHaq7YVXelrqrckhSza/CGmEsuQnOvlGamRvzvj3VkzGFHUtSLx2lFiO9lFS/5k7+2DPph9eB1e+ZkqKvqrvm5KkbkE6v3/KZw3SZDwGWAABC4Iwh0J83tvOCSmvsjCuY2bOveSFSTwlp4bvd9a/75KYdfKJeExh7wWbH6gX+Q5hYJ4p9YufnT85xMpSaTlraZaq6r11yZihP9xM8h3/GL74g+Ao0EAkDg9hHoNppLCu7YDdta3W4eaiDjuXlqts7I1iiSMs88v3L9DGpxXTV31q/r57ArGpCWRhz6ndDc0tA9vz8VLatqeIP4j/xcv+A9T/y8bSEbBLcZGngCBIBAewh0D81VO5yk5vZZsfoWNLdJPLfucuzR3x84zK6yZb++8p/99muBXdGF3Uhzr+jWbvztU2qL0AHS3D4rVt+3+Tylw+Nv7TEHHAMEgICHE+gemltXV9couxsupnfGJi3G0AjNHRrKrkJ+7ubvxI3aejnp5Ma2/dynTqWu3PbbuFC2qWFArjP1gXOAABC4Iwl0G829VdklYwtkrphSEZsZ8cTK9XPoxXV1lzNjtg/efPqowlFUWcouPPPED83juUlEPJdxqUk8V13lYM5e/9sbcTIiInxHdhxoNBAAAp0i0J0095Zkt8mciD4rVg/e8M/CHHIMra6uxhkbf+KFNesG/rz+yZBTyw4GDz14Xd7CH3vmpKmL6i4nnfjjqTiUn1ulyw5YvfnTAnunsMNJQAAI3KEEupnmNspuh+dE3KH2hWYDASDgXgS6n+aSsuteFKE2QAAIAIH2EeiWmtu+psFRQAAIAAG3IwCa63YmgQoBASDgwQRAcz3YuNA0IAAE3I4AaK7bmQQqBASAgAcTAM31YONC04AAEHA7AqC5bmcSqBAQAAIeTAA014ONC00DAkDA7QiA5rqdSaBCQAAIeDAB0FwPNi40DQgAAbcjAJrrdiaBCgEBIODBBEBzPdi40DQgAATcjgBortuZBCoEBICABxMAzfVg40LTgAAQcDsCoLluZxKoEBAAAh5MADTXg40LTQMCQMDtCIDmup1JoEJAAAh4MAHQXA82LjQNCAABtyMAmut2JoEKAQEg4MEEQHM92LjQNCAABNyOAGiu25kEKgQEgIAHEwDN9WDjQtOAABBwOwKguW5nEqgQEAACHkwANNeDjQtNAwJAwO0IgOa6nUmgQkAACHgwAdBcDzYuNA0IAAG3IwCa63YmgQoBASDgwQRAcz3YuNA0IAAE3I4AaK7bmQQqBASAgAcTAM31YONC04AAEHA7AqC5bmcSqBAQAAIeTAA014ONC00DAkDA7QiA5rqdSaBCQAAIeDAB0FwPNi40DQgAAbcjAJrrdiaBCgEBIODBBP51zS3860P/1h4BU2fOWb5m35lsTdnVlnwr5RdCfpmz5LCw9rq3Wh7qDs+drPDf5n+85KSmW9S2gVix/MLeX+b8EKlpeKE0+deA1izlPykgYNrM+au2hqVIyxoOvqX/ZcK4zUtnfrXXDeyrjfxxwap/zsuKu5Xtbok+nPwfE/jXNTf7j3db/SBfe3Hq4v159mYYZMfmTJrg//EO9uXu8Emo5eyYPtF/0ryTyu5QWwTahWfsWjJ1AjLBvOONmmtPWHnNKDfYm/ZDpKzylpvpRvbVnpg/CbV16uKd6UZXs14IT4DAv0KgizT3s+As3OpwNG4mjYydG3do3fxA4pMfMHtnYcm19rnRZ/JapW641800t4x1+NtAQlKnLd+bpqpubFe95s47JGg0E7lj0Mgo6WF/LiDP+iy44Fa9XXeyr0uVsW/FRwSPwKBQxq02rZEm7ACBGxDoKs3dw2u9Ag7avkVTUI+fsYNd3eBAudNnsvVqN3216zW3tkyWE3Uknt9xv8wlCFuGAgiTpn1/nNlCX+o1d2EE3rR1TfbV0SuR7E6ad6tRlK63702IlbHCfpyG/N2AoOO8jlNtwgh2gcDNCPzXmltXV2dIWPH2BH//wHWZDa5u138mb4aprfe7XHOrM/4I8PefsOpcC9Fsq5LEe47czYSyBH4fpbpeWW6quXW18sNzkDJ9eVx107LaOqDL7dsOYq6Gb5QZG7Mb+mFbbYD3gEAnCbiB5taVnf/5HX9//+k7G3zhLv9MdhIeeVrXa27qms5obnHehukT/f39P9vRenDg5ppbV8cM+dTf3//t1SkdlftmhLvcvtXtIlZWuH0W+sn18frcooafXM3qDU+AwG0g4A6aW8ffNxt9ktem1DtfXf6ZvCWQ3URzpYe/RoLy+Q72DQbB2qO56qPz/P39Jyw/67gVZF1u3/Zpbl1dJSfkC8KRPyi5lfbBuUCgDQLuobm7Cc39vT2a63II0sO2/jx/9vTAAH//KYEzF/4UEpmtqURtLBMkH/lz3UnsSmsNduH58ft+/37Ox4EBk9CJ02YvWRV8Ik3cPGWi8cxKE+tc6Ibl35DHB0xtSJaqbTyiYae55jrE6WHBP8//fFpggH8AUb2/TqTKHG36TbVlspSTf636ZubUgIBJ/gGB0+YsWYMa1SKLrjh51TvEkCMx4nPtz6TFUXib16+rq6vMW//BRBTAuWhrqHfL/+3R3Ho/t/HbsfEa7WwCeXwzzS3T5ESFrA66xvmH9UfOMR3Xxz4ay6qrq2tncR0n5kj+FcWsp67PrbgZ0qb1gX0g0G4C7qC5zvM/otjCZ43jbM0+k02aUiY//ccXreeQfvxTlEgdu+JNf/+AdZnXaa6VdqR+bPqaUjXsBc7fkdXCa3MUHFgyrTV18/cPXLI3x9T809iouVJjdshCcnC/4eIN/wPnbUtRtyojLm3GX/Pfbziu+f9pzbPoijPWfvB2AHoQh00idgMCAgKXx99Mc6uz/kAVm7aV2jhQ2YQruXtzza2VH56L3MBvmsdzO9AEsqQG+7Ks1CPLpjZvc/2zgM/XxIuv5VQ0rWwHiusEMRdtK4rABG7MrGpaKOwDgdtFwA00VxU+PwCNoW3Ma+jlDZ/JZvm5lcIjQUSGQ8C0JVtPpNGlGgOOq2SUlJN/Lf8IqdDUxUtmT2hFc00ZG2aiOKZ/wLQVwZHoRJPDYdLIKElhW5dOm+T/3u8pTTXXkfXnTGIIe9ryrWEphTIt7rDiGjErLXJbvRB/vpXaNN5Xr7kzViz/DJUydd6GQ+cpAhluIs/auWLG24SQTFuXaGhpNm3Cqg8IcQ+ctyH0PEWswa0OXCXIid+/6nPirEkztmW39Ezb+0u5SWHM7SgO+96fBa3qPnngTTVXE9uQt9A0E7kTTSDtO3XxinmojYFfrdl3Nlsg0+AmjYybG7f3lzlTSCaLo6TXyW4niqur6wgxV+FmNItnejCjCT/YBQK3jcB/rbmmvL/mTEaS9NUhWePP9lY0t4wZQoxvTPspSnTd57DOJYtePY0QtpaaW6uNXIaUOuDz9Uma60+sK1Oxm/3wNyWseG+Cv/+0n89IWhkmctD2zUe1nR7cRLxIzSVKD1hymHX9jCaX9vwfX6D3pyyP1zfxkWvlJ4g8uYB5OyjWJq+Txm08a+pPSZZm73ZEQchrmSKD3vT39/8+saV8N+1H9Zp7fX6uSSOjp4dtXRCIvor8p29tMgTXuSaQ9kUXC/gyOAtvtHtjbRp6RcC8w9d6BQopdAWxeg5Lbpgz11hN2AECnSDQVZrb6pyIvevqPZopi082VdLrNdcQveTtCf6TZuyk3SiPp6zwL0KUm8cWyrI2o9/UU5ZHtWtirov0Bz/bcePceNmx+W9O8J/yU5K9QQcbNTewpThes4dLSP4qnx7MaPQ0SzOI3/uBPyXdKDLgkh+ej74wrkVdiCt2WHNrOcTv5Wk7WddFXa5Vsa5ea4gvjxv9mbaq2Ty0TjahQXMDlkRorhdcskra6CUoeB24Lu2axTtZXMf83Lo6zh70/T1ta7OfWU1AwS4QuBUCXaS5N/oMI1dnxk9R/NJmbbhOc9Xhi9GRP7Q5XG6IRmrYTHPLUn9FM48/29+QhdasmOueuGhbP5jof5MhKXK2aODG7IZISIPmtlDGFlev14trn+Sy87+iKPaXbQ6R139nNJ8G3WHNdWWsRd9YN5md3KbmBs5ZveP0JXVz37+zTajX3CYMW8BCT12N43UNhXa2uI5qriocdaSANek3jn23Ul94CQi0j8B/pLmTAgI/noOWTUli4o2OX2ONW2puWeo6JJ1BUdfFQxtPIX54Ehn7TcbQ6v27Gftbz2RoejKx3+B/tfEN0fjW93ENv9PrNTdwW2FbXmRdcfIqQviiyPBCY934bZ5VlrH2HWIxhyZBiQ5rblnzoq9rN/lCvea2mIdWduOEh043geQcsCa9tOG3QmtVchUQv1Fm7KlfCqfTxXVUc8kv74CfktqsXmtVhteAwM0JdJHmtu0D [... cut ...] YV7hiVIgkSVYEAjSAAABN0lEQVTc/XXDIiFcXr8g2AYCAiYgXOcKOFmEAwEQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIgDnilVxIVkQAAExJwDninkBIn0QAAGxIvD//Z6XM5p0TUUAAAAASUVORK5CYII=" alt="image" iscopyblocked="false">

UsamaEquinorAFK, 2023-08-03:
Identified improvement while working on this PR #930

aestene, 2023-10-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

oysand, 2023-10-02:
The disabling of the localization button will not be needed after #950

closed 2023-10-02 by oysand