equinor/flotilla #939: Make areaName optional for mission scheduling

oysand, 2023-08-04:
**Describe the improvement you would like to see**
Scheduling a mission should not require the areaName.

**How will this change existing functionality?**
Currently the areaName is required and will fail the mission scheduling. The area should be determined based on the TAG's in the mission.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-08-04:
[image]

andchiind, 2023-08-07:
I think this can be closed, right?

anetteu, 2023-08-15:
@oysand closing issue due to lack of detailed description

closed 2023-08-15 by anetteu