equinor/flotilla #955: Schedule missions based on events rather than a scheduler loop

aestene, 2023-08-08:
**Describe the new feature you would like to see**
The scheduling of missions should happen on an event-basis rather than as a scheduler-loop which is continuously checking if there is a new mission to run.

**Describe the solution you'd like**
Currently, the mission queue is continuously monitored by the mission scheduler which automatically schedules a mission if there is an available robot and the time of the mission has arrived. 
1. We no longer require the support for timing missions
2. This loop is causing several timing issues with other parts of the application that is the source of bugs

Instead, we would like to move towards a more event-based system where the missions are scheduled based on events. There are mainly three events which should be incorporated:
1. New mission event
    - A new mission has been scheduled and an event should be triggered. 
      - If there is no mission running and the robot is available the mission should be scheduled through ISAR
      - If there is a mission running the mission should be added to the queue for that robot
      - If the stop all robots field is active the mission should be added to the queue but it will not be scheduled (see below)
2. Mission finished event
    - A mission has finished and this event is raised by an MQTT message from ISAR which says the robot is now `Available`
      - The robot is set to `Available`
      - There should at this point be no mission running so a new mission is pulled from the queue if there is one and scheduled
      - No missions are scheduled if the stop all robots field is true
3. Stop all robots event
    - The button to stop all robots have been pressed 
      - The current mission should be cancelled and added to the top of the mission queue for that robot such that it may be resumed at a later point
      - A mission to return the robot to its safe position is scheduled immediately
      - The field for "stop all robots" is set such that no other missions may be scheduled until the operator manually clicks the button again which should clear this field 

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

Afonso-2403, 2023-08-11:
Adding on to the description above, there should also be a fourth event which would be:
4. Resume all robots event
- The button to resume all robots has been pressed
  - The field for "stop all robots" is set to false
  - If the mission to return the robot to a safe position is running it will trigger a Mission finished event which leads to fetching new missions from the queue
  - If no mission is running, the robot is set to `Available` and a new mission is fetched from the queue and scheduled

aestene, 2023-08-22:
Partly resolved by #973 except stop button event is being solved in a future PR.

oysand, 2023-09-21:
@aestene Do we have an issue to address the final functionality?

aestene, 2023-10-30:
Closing as remaining functionality is implemented through #1045

closed 2023-10-30 by aestene