equinor/flotilla #980: Robot objects should be valid after call to constructor

aestene, 2023-08-17:
**Describe the improvement you would like to see**
Instantiating a `Robot` object using the constructors for the class should be sufficient to have a valid object which may be stored in the database. 

**How will this change existing functionality?**
The `Robot` object is stored in the database and has a `Required` field `RobotModel`. This field is not included in the `CreateRobotQuery` and is not addressed in the default constructor for the `Robot` class. Instead it is added afterwards based on a database query to find the correct `RobotModel` which matches the `RobotType`. 

As the `RobotType` is not stored as part of the `Robot` object, once the constructor call is finished there is no way of knowing which `RobotModel` is linked to it. Currently, the solution has been to add the `RobotModel` to the `Robot` object wherever it is required immediately while the `RobotType` is available.  For example the way it is handled in `MqttEventHandler` whenever a new `Robot` is discovered.
https://github.com/equinor/flotilla/blob/97904df7bc2efbd92f9113f82ce0e2c40da09753/backend/api/EventHandlers/MqttEventHandler.cs#L119-L146

Here the `RobotModel` is read and added to the `Robot` object before `RobotType` goes out of scope. 

**Desired solution**
Making a call to the constructor for the `Robot` class should be enough to ensure that the `RobotModel` may be included when the `Robot` is stored in the database. One solution could be to add the `RobotType` as a field on the `Robot` such that the `RobotModel` may be queried and added to the `Robot` object when calling `RobotService.Create`.

Alternatively, it might be possible to query the database directly for the `RobotModel` in the constructor although this required including the `RobotService` in the `Robot` class which might be a bad pattern. 

**How will this improvement affect the current Threat Model?**
Shouldn't have any particular effect.

andchiind, 2023-10-12:
@Christdej can we close this?

Christdej, 2023-10-12:
Yeah, looks like my previous Pull Request solved this

closed 2023-10-12 by Christdej