equinor/flotilla #986: Manage access to Flotilla through user access groups in Azure AD

aestene, 2023-08-22:
**Describe the new feature you would like to see**
It is required to limit access to robots and functionality based on the facility and the role of the person working on it.

**Describe the solution you'd like**
Suggestion is to create some initial groups to see how well it works and then build from there. Currently we have mission control and admin as groups. We would like to expand upon that to have both mission control and admin groups for each installation.
- Begin with the main installations that we are looking at
- The solution should use AccessIT to manage access to the groups

**How will this feature affect the current Threat Model?**
This should improve the threat model by making management of access easier and clearer.

UsamaEquinorAFK, 2023-10-26:
Made a request to add Flotilla to accessIT, the ticket is up and added Øystein and Arnt Erik in the watchlist. @oysand

UsamaEquinorAFK, 2023-10-30:
[image]

UsamaEquinorAFK, 2023-10-30:
Right now it is setup like this. @aestene @oysand

UsamaEquinorAFK, 2023-10-31:
@aestene @oysand - anyaction here should be close it or do anything further ? Put it in blocked ?

oysand, 2024-01-08:
Completed

closed 2024-01-08 by oysand