equinor/flotilla #998: Only schedule mission on the same deck at once

oysand, 2023-09-05:


andchiind, 2023-09-19:
I'm confused. Are you saying we should both clear missions on other deck but also confirm the robot on this new deck? Or is this issue that the user should be notified if they attempt to queue missions for a deck the robot is not currently localised on and that these missions should not be put on the queue? And you then want to clear the queue if the robot is localised again on a different deck?

oysand, 2023-09-20:
This is meant make sure that when we ask the user to confirm that the robot is placed at the correct deck, it is only meaning one deck. This issue is two intertwined with the other issues to split up. Will therefore close it and try to write some new ones

closed 2023-09-20 by oysand