equinor/flotilla #1010: Create an integration test for running a mission

andchiind, 2023-09-18:
**Describe the improvement you would like to see**
Create an integration test which checks that that we are able to create and start a custom and echo mission. This test should utilise a database that is not in memory, such as the dev database but ideally we would use a test database. We should also not mock anything, but we should use isar-robot as our target robot. This test should run as a github action check before merging a PR, but ideally after finishing migrations.

**How will this change existing functionality?**
This will allow us to test that there won't be any significant errors that halt testing offshore, and it will let us detect database validation errors which the in-memory database is unable to simulate.

**How will this improvement affect the current Threat Model?**
Assuming we correctly retrieve our keys for validating our HTTP requests in the github actions, the threat model should not be affected.

oysand, 2023-09-25:
Check if we can use the test API for STID, SLIMM and Echo

Christdej, 2023-11-21:
To be able to call the backend from the from the frontend we have a few choices:
- Make a testuser that can have access to test versions of the integrations we have (STID SLIMM ECHO).
- Make it possible to disable authentication on the backend and do this when testing
- Mock the authentication when testing for both frontend and backend

Christdej, 2023-11-21:
The easiest option might be to set the environment to "test", and then make the Program.cs use the same authentication and authrization setup as the TestWebApplicationFactory when the environment is "test" and then in the frontend create a test-user that will be able to authenticate.

UsamaEquinorAFK, 2023-11-23:
Have we fixed the authentication problem ? @Christdej @andchiind ?

Christdej, 2023-11-24:
Nope,
I would suggest doing 
#1203  and #1204 
Before starting on this

aestene, 2024-01-27:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2025-01-13:
Related to change with tests, were the tests will not share database resources. Also easier setup of tests.