equinor/flotilla #1019: Evaluate battery and pressure levels before starting missions

anetteu, 2023-09-22:
**Describe the new feature you would like to see**
It should be clearly displayed if the battery and pressure levels are sufficient to start a mission or not. Consider a warning in the frontend when battery/pressure levels are starting to get low (#722). Furthermore, an error message, telling the user that the robot must be charged and/or pressurized before starting a mission, should be considered. 

**Describe the solution you'd like**
A suggestion could be a pop-up explaining the required actions by the user before it is allowed to start a mission. 

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-09-25:
Lets keep the scope to pressure for the first implementations

aestene, 2023-11-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2023-12-20:
Awaiting PR which reworks the update robot endpoint. It is currently broken.

closed 2024-01-30 by andchiind