equinor/flotilla #1024: Improve status message for arm movements

anetteu, 2023-09-23:
**Describe the improvement you would like to see**
The status message should be removed or redesigned. 

Current display: 
[image]

**How will this change existing functionality?**
A clear and concise description on how this will affect current functionality.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-09-25:
Align with UX

aestene, 2023-11-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2024-01-18:
#1338

closed 2024-01-19 by Eddasol