equinor/flotilla #1025: Implement behaviour when robot is not capable of continuing a mission

anetteu, 2023-09-23:
**Describe the new feature you would like to see**
For the each robot, it should be defined when

- The robot is capable of starting a mission (covered in #1019)
- The robot is capable of continuing a mission (starting a new inspection) (this issue)

Such a definition shall include

- The values that are affecting the capability of the robot to continue a mission, and at what thresholds the robot is considered capable/not capable (examples are battery level and pressure).
- The expected behaviour when the robot is considered not capable of continuing a mission.
- If the capability of the robot to continuing the mission is evaluated continously, or only at the beginning of each inspection point (i suggest the latter).

**Describe the solution you'd like**
An example could be for the robot to go to home or safe position when battery and/or pressure level is below some threshold. Neccessary information must be displayed in the user interface. 

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-09-25:
Lets keep the scope to pressure for the first implementations

aestene, 2023-11-25:
This issue has automatically been marked as stale as there has been no activity for 60 days.

closed 2024-07-31 by Eddasol

Christdej, 2025-01-15:
Robot returns to dock when battery is below threshold and does not start a new one before battery is above the start mission threshold