equinor/flotilla #1026: Bug in changing pressure display in mission page depending on pressure status.

UsamaEquinorAFK, 2023-09-25:
**Describe the bug**

The feature request implemented in #886 is not working in prod. 

**To Reproduce**
Works fine in dev on placebot. Test on isar-taurob to debug. 

**Expected behavior**
Different colors for various pressure values. 
[image]
[image]

**Screenshots**
Currently in prod it looks like this which is incorrect. 
[image]

UsamaEquinorAFK, 2023-09-25:
@anetteu - Is it phrased correctly ? :)

aestene, 2023-12-18:
Was working on previous test with Porthos

closed 2023-12-18 by aestene