equinor/flotilla #1035: Remove warnings in broker

Afonso-2403, 2023-10-02:
**Describe the improvement you would like to see**
Upon start, the broker emits a warning that the [passwd_file](https://github.com/equinor/flotilla/blob/main/broker/mosquitto/config/passwd_file) and [access_control](https://github.com/equinor/flotilla/blob/main/broker/mosquitto/config/access_control) files are not owned by mosquitto and that future versions will refuse to load this file.
There is a suggestion on how to fix this, it should be tested and maybe added to the DockerFile? 
[image]

**How will this change existing functionality?**
There will be no change to functionality, just preventing issues in future updates of mosquitto.

aestene, 2023-12-02:
This issue has automatically been marked as stale as there has been no activity for 60 days.

mrica-equinor, 2025-07-07:
Fixed by #2250