equinor/flotilla #1046: Improve start of missions from the mission queue

mrica-equinor, 2023-10-05:
**Describe the improvement you would like to see**

Currently the mission queue attempts to start a mission regardless of checking that the robot has been localized in the correct place and that will respond correctly in an emergency. This mission is sent to the robot and it fails because the robot can't reach the robot poses in that mission.

Before sending the mission to isar, the mission queue should check if:

- The robot is localized in the same deck as the mission
- At least one area in the deck has a safe zone to which the robot can go to in an emergency state

For now it's okay to assume that each deck will have portals between all the areas so it's free to move in the same deck and only one safe position per deck is necessary.

**How will this change existing functionality?**
The backend will have more control over what missions are being sent and started in isar.

**How will this improvement affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

oysand, 2023-10-16:
To be addressed through #950 except safe position requirement

Laxynium, 2023-10-23:
Hello, I would like to start working on this one. I've already checked out a repository and I've tried to run backend Api.
I've managed to make it working with my own azure tenant, however there are also some services(represented in code as IDownstreamApi) on which this api depends, which I was not able to fake. 
Also I didn't managed to make it working with mosquitto.
Could you guide me what is needed to complete this issue, since I'm not sure if I can complete it without access to yours tenant on azure or without some another configured dependency?

mrica-equinor, 2023-10-23:
> Hello, I would like to start working on this one. I've already checked out a repository and I've tried to run backend Api. I've managed to make it working with my own azure tenant, however there are [...]

Hei, unfortunately we don't have the code ready to be able to have open source contributions from people who do not have access to the downstream api services we are using. This is a case that we hadn't encountered up until this point and we have created an issue to address this so that people can contribute in the future. #1079 

We really appreciate your interest and hope that you and other will be able to contribute very soon :)