equinor/flotilla #1061: Enforce correct relation to installation for new robots

aestene, 2023-10-14:
**Describe the improvement you would like to see**
Whenever a new robot is added to the database this is done based on discovery from the MQTT messages being sent from that ISAR instance. These messages contain a string which is the installation code that the robot is on. This string is put straight into the `Robot` model and not verified again existing installations. The side effect being that you may have rogue robots which are not properly linked to an existing installation in the database. 

The robot should instead link to an installation properly in the model and not just to a string installation code. Then the robot will be verified when connecting that it belongs to an installation which exists. 

**How will this change existing functionality?**
Improve stability and verification.

**How will this improvement affect the current Threat Model?**
N/A

ssingh3856, 2023-10-16:
Can you please fill me on where exactly to look? Was just going through the codebase but I from the description I could not really locate the files to be modified

aestene, 2023-12-16:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2024-01-03:
This has already been added in [#1297](https://github.com/equinor/flotilla/commit/7cb858f5445f9920c6385def3907982975045815)

closed 2024-01-03 by andchiind