equinor/flotilla #1077: Improve logging in frontend

Christdej, 2023-10-23:
**Describe the improvement you would like to see**
At the moment we only do console logging in the frontend. We should change this to a logger so the logs can be sent to application insights and be seen in the CICD system

**How will this change existing functionality?**
Make it easier to degbug frontend bugs in our environments

andchiind, 2023-12-27:
Relevant information: https://learn.microsoft.com/en-us/azure/azure-monitor/app/javascript-sdk?tabs=npmpackage

UsamaEquinorAFK, 2024-01-09:
Closed: https://github.com/equinor/flotilla/pull/1320

closed 2024-01-09 by UsamaEquinorAFK

UsamaEquinorAFK, 2024-01-09:
Robotics Infrastructure: 
https://github.com/equinor/robotics-infrastructure/pull/379
https://github.com/equinor/robotics-infrastructure/pull/380