equinor/flotilla #1078: Detect if a person is present in an image prior to uploading it.

oysand, 2023-10-23:
**Describe the new feature you would like to see**
Prior to uploading any normal images or videos the system should check if there are people present in the image and only upload if there are no people present to adhere to GDPR requirements.

**Describe the solution you'd like**
If the inspection data contain a person, the step should fail with this as a reason.

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-12-26:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2024-02-12:
Blocked waiting for computer vision solution to be matured / developed

UsamaEquinorAFK, 2024-02-23:
Do we have any next workshop planned with them ?

tsundvoll, 2024-04-29:
We have an example now on how this can be done; ready to be tried out. Not really a Flotilla issue, will be handled by the storage / analytics.

Eddasol, 2024-05-02:
Will be covered by https://github.com/equinor/analytics-infrastructure/issues/7

closed 2024-05-02 by Eddasol