equinor/flotilla #1079: Allow for open source contributions to flotilla

mrica-equinor, 2023-10-23:
**Describe the new feature you would like to see**
We have been contacted by someone who was interested in contributing to flotilla by completing one of the cards in the board. Although this person managed to make it work with their own azure tenant, some services like Echo (called IDownstreamApi) can't be accessed because we have not implemented the necessary resources/mocks to allow for open source development from people outside of the team.

**Describe the solution you'd like**
We already have the custom mission endpoint which could allow for people to contribute without access to echo, but we need to refactor the code so it will be able to run without the credentials necessary to connect to echo.

**How will this feature affect the current Threat Model?**
No features should be affected by this improvement.

Christdej, 2023-11-21:
These 
https://github.com/equinor/flotilla/issues/1203
https://github.com/equinor/flotilla/issues/1204
Needs to be done to solve this

aestene, 2024-01-21:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2024-05-02:
Covered by #1575

closed 2024-05-02 by Eddasol