equinor/flotilla #1087: Show warning or message in flotilla if actions is needed

anetteu, 2023-10-27:
**Describe the new feature you would like to see**
It should be very clear for the users if any action is needed (before starting a mission). Such messages should be written in text. An example can be: "The pressure on robot is below xx mBar. The robot must be pressurized before starting any missions."

Related issues #722 #1025

**Describe the solution you'd like**
The message can be a general pop up or something connected to the robot card. 

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

aestene, 2023-12-27:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2024-03-25:
@anetteu Can this be closed now that we have warnings before adding missions to the queue? We have https://github.com/equinor/flotilla/issues/1473 which addresses stopping the queue if the status changes again

anetteu, 2024-04-15:
I cannot see those warnings? @andchiind

Eddasol, 2024-05-02:
Solved in https://github.com/equinor/flotilla/pull/1372

closed 2024-05-02 by Eddasol