equinor/flotilla #1088: Action log in flotilla?

anetteu, 2023-10-28:
Something similar to what is in permit vision, along the lines: 

Mission A,B,C on Area 1 scheduled by Operator X at hh:mm dd.mm.yyyy
Mission A started on Area 1 at hh:mm dd.mm.yyyy
Mission A completed 
Mission B started on Area 1 at hh:mm dd.mm.yyyy
Mission B on Area 1 stopped by Operator Y at hh:mm dd.mm.yyyy
Mission B and C at Area 1 cancelled by Operator Y at hh:mm dd.mm.yyyy

Can include only actions done by Operators or also start/conplete/fail/stuck? actions by robot. 

Just an idea, open for discussion

oysand, 2023-10-30:
Remember GDPR if we implement this

aestene, 2023-12-30:
This issue has automatically been marked as stale as there has been no activity for 60 days.

Eddasol, 2024-05-02:
Covered by https://github.com/equinor/flotilla/issues/1569

closed 2024-05-02 by Eddasol