equinor/flotilla #1099: Add robot instructions and manual to flotilla

anetteu, 2023-10-31:
**Describe the new feature you would like to see**
Make relevant instructions, manuals and documentation of the robots available in flotilla. 

**Describe the solution you'd like**
A suggestion could be an info button on the front page or on the robot page that will lead to a page or popup with the relevant documentation.

**How will this feature affect the current Threat Model?**
Leave blank for maintainers to fill out if you are uncertain about this.

tsundvoll, 2023-11-06:
Be aware of access-restrictions to this information. Link this from blob-storage.

aestene, 2024-01-06:
This issue has automatically been marked as stale as there has been no activity for 60 days.

prasm313, 2024-06-14:
Talked about in UX meeting 13.06. UX designer has made a suggestion to how this could look in the robot page.

We want revisions to be handled in STID or Docmap. We should therefore use STID links placed in Blob or Azure.

Eddasol, 2024-08-15:
Design started in PR
https://github.com/equinor/flotilla/pull/1721

closed 2024-09-13 by Eddasol