equinor/flotilla #1101: Implement error recovery functionality for event handlers

aestene, 2023-11-01:
**Describe the new feature you would like to see**
Currently in most of our event handlers we have empty return statements whenever there is a problem that arises. For example upon reading something from the database, if it returns null we log the problem and then return. However, it might be we should perform some error recovery actions.

An example is in the `OnMissionRunCreated` event. If the `missionRun` variable is null we return without handling anything. It is likely that we have a rogue mission run somewhere here and maybe it should be handled? Or maybe it should be flagged higher than with just a log statement.

https://github.com/equinor/flotilla/blob/59462a5dd1787d9cc1558645171f84577e62c39a/backend/api/EventHandlers/MissionEventHandler.cs#L72-L93

**Describe the solution you'd like**
Investigate if we should implement more error recovery behavior for event handlers. This is an open issue where investigating if it is required is at the core. It might be the solution is to flag the event as they are very rare and use them as indicators to find bugs. 

**How will this feature affect the current Threat Model?**
N/A

aestene, 2024-01-06:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2024-08-12:
Although I have added try/catch to almost all places where we may get an exception, it is worth noting that it is possible to get an exception on all database calls, despite this being rare. We may want to have a systematic solution for this, either at the service level, event handler level or even in the ASP.net setup.

closed 2025-07-07 by mrica-equinor