equinor/flotilla #1130: Create a test for RobotStatus Enum

Afonso-2403, 2023-11-08:
**Describe the improvement you would like to see**
The [RobotStatus](https://github.com/equinor/flotilla/blob/20b632e6931d82cc8bdf748af9821910d34bfdf3/backend/api/Database/Models/Robot.cs#L116C20-L116C20) Enum in the backend needs to have the same ordering as the [RobotStatus](https://github.com/equinor/flotilla/blob/20b632e6931d82cc8bdf748af9821910d34bfdf3/frontend/src/models/Robot.ts#L7C2-L7C2) Enum in the frontend, since in the conversion it is interpreted [as a number](https://github.com/equinor/flotilla/blob/20b632e6931d82cc8bdf748af9821910d34bfdf3/frontend/src/components/Contexts/RobotContext.tsx#L31C29-L31C29).
We should have a test that checks that the conversion between Enums is as expected, so that if a new status is added in the wrong place we can immediately catch the bug.

tsundvoll, 2023-11-27:
Consider converting Enums to strings

Afonso-2403, 2023-12-18:
Not relevant once #1272 is implemented