equinor/flotilla #1154: Implement Role Based Access to Flotilla

Afonso-2403, 2023-11-09:
**Describe the new feature you would like to see**
Users should only be able to interact with what they have access to (operators at Johan Sverdrup should not be able to see robots at Kårstø, for example).

**Describe the solution you'd like**
Ideally this would be through giving a specific role (Reader JS, for example) to a given group in Azure (Operators JS, for exemple). This way, new operators will automatically get access to Flotilla and operators leaving will lose access once they are added/removed to such a group.

**How will this feature affect the current Threat Model?**
This will prevent information disclosure between plants and improve elevation of privilege protection.

aestene, 2023-11-13:
Related to #986

UsamaEquinorAFK, 2024-01-02:
@andchiind should this be closed now ?

oysand, 2024-01-08:
Completed

closed 2024-01-08 by oysand