equinor/flotilla #1188: Host SignalR locally for development

andchiind, 2023-11-16:
**Describe the improvement you would like to see**
SignalR should be hosted or emulated locally when the in-memory database is selected. This will prevent message conflicts when testing entirely locally. Two potential options are the local emulator: https://github.com/Azure/azure-signalr/blob/dev/docs/emulator.md and the self hosted server: https://learn.microsoft.com/en-us/aspnet/signalr/overview/deployment/tutorial-signalr-self-host . The chosen option will need to be started in Program.cs when the in-memory database option is selected. 

**How will this change existing functionality?**
Conflicts in messages when developing with the in-memory database will be prevented.

**How will this improvement affect the current Threat Model?**
N/A since staging and prod will be unaffected, as will the dev database.

tsundvoll, 2023-11-27:
Investigate if maybe we can use SignalR groups instead

andchiind, 2023-12-20:
This is no longer needed due to the use of signalR groups