equinor/flotilla #1195: Replace semaphores in scoped services with a shared semaphore in a hosted service

andchiind, 2023-11-17:
**Describe the bug**
Semaphores are being ignored since they are local to the current scoped service instance. Each time a different dependency injection is done, a new semaphore is created.

**To Reproduce**
Set the safezone status off and on several times while debugging. Updating the robot state is not truly being protected by a mutex.

**Expected behavior**
A semaphore should block any thread from accessing the protected code region. This can be done by storing them in hosted services which only have one instance at a time (as long as flotilla only uses one server at a time).

**Screenshots**

andchiind, 2023-12-13:
A singleton might also work instead of a hosted service

andchiind, 2024-01-03:
The only remaining problematic Semaphore is in RobotService.cs

oysand, 2024-01-08:
Consider removing the semaphore

UsamaEquinorAFK, 2024-01-09:
https://github.com/equinor/flotilla/pull/1323

closed 2024-01-12 by aestene