equinor/flotilla #1204: Make it possible to disable authentication and authorization for only running the backend

Christdej, 2023-11-21:
**Describe the improvement you would like to see**
This will make it easier to run for people outside Equinor and easier to run in integration tests.

**How will this improvement affect the current Threat Model?**
If set up wrong can increase the risk of starting with app without authorization and that can give unwanted access to resources.

aestene, 2024-01-21:
This issue has automatically been marked as stale as there has been no activity for 60 days.

tsundvoll, 2024-01-22:
Consider implementing / providing an openly available authentication provider with documentation for how to use it. In order for open source contributions to take place.

Christdej, 2025-07-07:
Can be done more or less how it is done in:
https://github.com/equinor/flotilla/tree/disableauthconf

closed 2025-07-07 by Christdej