equinor/flotilla #1220: Remove default Robot constructor

tsundvoll, 2023-11-24:
**Describe the improvement you would like to see**
The default robot constructor in `Robot.cs` is not really in use and can be removed

**How will this change existing functionality?**
Not much, but will make the code more readable

**How will this improvement affect the current Threat Model?**
N/A

UsamaEquinorAFK, 2023-12-07:
What do you mean ? Its being used in initdb.cs I think (snap attched below).

[image]

andchiind, 2023-12-14:
I tried removing it, but entity framwork seemed to want to have a default constructor 

[image]

I am also not sure if this needs to be prioritised, so I'm moving it to TODO

andchiind, 2024-01-02:
@tsundvoll I hope you don't mind me closing this issue for the above reasons

closed 2024-01-02 by andchiind