equinor/flotilla #1226: Requests should return 400 if they fail to schedule a mission

andchiind, 2023-11-28:
**Describe the bug**
When scheduling a custom mission, if we attempt to schedule it in an area the requested robot is not in, it will state that it failed and instead return a 200 code, with a mission run with a null area.

**To Reproduce**
POST a custom mission in an area with no robots

**Expected behavior**
This should return a 400 error stating that there are no robots in the area (even if there are robots there and they just don't have access, since we would otherwise be giving them information they do not have access to ie. that there are robots there)

**Screenshots**
If applicable, add screenshots to help explain your problem.

oysand, 2023-12-11:
Should be handed as part of #950

Eddasol, 2024-04-18:
Tried to reproduce but doesn't occur as described anymore. Will close

closed 2024-04-18 by Eddasol