equinor/flotilla #1227: Queue is wrong after pressing "Queue all missions"

Eddasol, 2023-11-28:
**Describe the bug**
After queuing all missions for a deck the queue is in the order the missions completed loading and not in the order it will run. Reloading will show the correct order.

**To Reproduce**
Recommend starting a mission and pausing it before doing the following steps such that the missions don't start instantly

1. Queue all mission on a deck with several missions
2. Wait for all the missions to load, note the order
3. Reload the page, note the order

**Expected behavior**
The queue order should match the order the missions will run in

**Screenshots**
After queuing all missions
[image]

After reloading
[image]

andchiind, 2023-12-08:
The reason for this is that each mission does not know its ordering. The ordering currently is arbitrary on the backend. So when the signalR message is sent we don't know how it has been ordered on the backend. We might need to consider implementing a consistent ordering, eg. having a single increment for each mission run that is globally unique

aestene, 2024-05-21:
This issue has automatically been marked as stale as there has been no activity for 60 days.

andchiind, 2025-02-13:
If we moved the mission queue to ISAR, we could achieve a consistent, global ordering. Currently this is not the case.

Eddasol, 2025-06-10:
Sufficiently solved by #2212

closed 2025-06-10 by Eddasol