equinor/flotilla #1228: Mission queue should be sorted by closest deadline

Eddasol, 2023-11-28:
**Describe the improvement you would like to see**
Missions on a deck is sorted by deadline and when scheduling all missions on a deck the queue should get the same order such that missions with a short deadline ends up first in the queue.

Missions on the deck, ordered by deadline
[image]

Mission Queue after queuing all missions
[image]

**How will this change existing functionality?**
Missions with short deadline will run first

**How will this improvement affect the current Threat Model?**
N/A

andchiind, 2023-12-11:
This could be solved by having global ordering, where each mission is assigned a unique incrementing number. This way we could allow the user to reorder the queue in any way they want, while also having it be predictable for a user. This would also prevent starvation where missions without a deadline are always put to the end of the queue.

andchiind, 2024-01-02:
It is hard to actually achieve the same ordering on the frontend and backend unless we purposefully slow down the frontend. This would also still not guarantee the same ordering. See the above comment for a potential fix.

andchiind, 2024-04-02:
Alternatively we'll need to ensure we use the same sorting of mission runs on the frontend and backend in respect to deadlines, but if two missions have the same timings, or if neither has an inspection frequency, then there would still be inconsistencies.

andchiind, 2025-02-13:
See https://github.com/equinor/flotilla/issues/1227#issuecomment-2656078422

Eddasol, 2025-06-10:
This feature (inspection plan and deadline) isn't used as much anymore and is therefor not as big of an issue. Suggest closing as not planned