equinor/flotilla #1252: False MSAL warning

tsundvoll, 2023-12-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Start broker with `docker compose up broker`
2. Run backend with `dotnet run --project api`
3. Start frontend with `npm start`
4. Observe the logs

```
2023-12-12 09:16:00 - warn: Microsoft.Identity.Web.TokenAcquisition[0]
      => SpanId:b61c0b92ea8e183b, TraceId:e1c3834f3d1bf9c0e0d063372bfc1a37, ParentId:0000000000000000 => ConnectionId:0HMVQT9NF3RTT => RequestPath:/echo/plants RequestId:0HMVQT9NF3RTT:00000037 => Api.Controllers.EchoController.GetEchoPlantInfos (api)
      False MSAL 4.57.0.0 MSAL.NetCore .NET 8.0.0 Ubuntu 20.04.6 LTS [2023-12-12 08:16:00Z] Only in-memory caching is used. The cache is not persisted and will be lost if the machine is restarted. It also does not scale for a web app or web API, where the number of users can grow large. In production, web apps and web APIs should use distributed caching like Redis. See https://aka.ms/msal-net-cca-token-cache-serialization

2023-12-12 09:16:01 - warn: Microsoft.Identity.Web.TokenAcquisition[0]
      => SpanId:80665597bfc4eb5c, TraceId:60290276817c8332e6954c63318cec53, ParentId:0000000000000000 => ConnectionId:0HMVQT9NF3RTT => RequestPath:/echo/plants RequestId:0HMVQT9NF3RTT:0000003F => Api.Controllers.EchoController.GetEchoPlantInfos (api)
      False MSAL 4.57.0.0 MSAL.NetCore .NET 8.0.0 Ubuntu 20.04.6 LTS [2023-12-12 08:16:01Z] Only in-memory caching is used. The cache is not persisted and will be lost if the machine is restarted. It also does not scale for a web app or web API, where the number of users can grow large. In production, web apps and web APIs should use distributed caching like Redis. See https://aka.ms/msal-net-cca-token-cache-serialization
```

**Expected behavior**
Expected no warnings

**Screenshots**
If applicable, add screenshots to help explain your problem.

UsamaEquinorAFK, 2023-12-29:
Cant recreate this in windows. Although I get this warning. Warning: File mosquitto/config/passwd_file has world readable permissions. Future versions will refuse to load this file.

sondreo, 2024-03-22:
The warning, which originates from MSAL for dotnet, suggests to leverage a distributed token cache.

For reference, a [distributed token cache scheme](https://learn.microsoft.com/en-us/entra/msal/dotnet/how-to/token-cache-serialization?tabs=aspnetcore) may be implemented, for instance by setting up a Redis cache in Docker. A distributed cache will persist, and a cached token will not be lost on server restart.

This warning should be considered developer feedback, and is a recommendation for web apps, and not necessarily a bug.

As a future improvement, we should consider implementing a distributed L2 cache for Flotilla.

closed 2024-08-05 by Christdej